sparkfun/SparkFun_Displacement_Sensor_Arduino_Library #1: [Bug] Invalid conversion from 'uint8_t' to 'ADS_DEV_IDS_T'

HarrisonLavins, 2019-06-23:
### Subject of the issue
When verifying/compiling example sketches in Arduino IDE 1.8.9 the code does not compile with _**error message: invalid conversion from 'uint8_t {aka unsigned char}' to 'ADS_DEV_IDS_T'**_

### Your workbench
* What platform are you using?
MacOS Mojave 10.14.3
* What version of the device are you using? Is there a firmware version?
SparkFun nrf52840 mini 
* How is the device wired to your platform?
N/A
* How is everything being powered?
N/A
* Are there any additional details that may help us help you?
I am using Adafruit nRF52 version 0.10.1 board drivers. See attached photos and text file with verbose compiler output for more info.

[Issue0_Output.docx](https://github.com/sparkfun/SparkFun_Displacement_Sensor_Arduino_Library/files/3317442/Issue0_Output.docx)

[image]
[image]

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.
Verifying any of the provided example sketches (i.e. Example1_1Axis_BasicReadings, Example2_TestDeviceType) with the Arduino IDE will result in compiler error messages. The issue appears to derive from the SparkFun_Displacement_Sensor_Arduino_Library.cpp file on line 57. _**Simply casting the result from readDeviceType() to type 'ADS_DEV_IDS_T' seemed to make the compiler happy and resolve the issue.**_

### Expected behaviour
Working code, and a happy user.

### Actual behaviour
Compiler errors and a sad user.

nseidle, 2020-01-22:
Sorry! It seems the Arduino compiler changed it's permissiveness at some point, or else I never had compilable code... Fixed now in v1.2.0.

closed 2020-01-22 by nseidle

skg91, 2020-06-15:
I am getting same error while compiling on arduino DUE. "Arduino\libraries\SparkFun_Angular_Displacement_Sensor_Arduino_Library-1.1.1\src\SparkFun_Displacement_Sensor_Arduino_Library.cpp:57:31: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'ADS_DEV_IDS_T' [-fpermissive]

   axisAmount = readDeviceType(); //Set global sensor axis number
"

Any solution?