sparkfun/SparkFun_Displacement_Sensor_Arduino_Library #2: 2 sensors on the same I2C bus, merged data

sideridispas, 2019-09-13:
### Subject of the issue
I want to read angle position of two 1-axis sensors, wired to the same I2C bus. Your example works great (after changing the default addresses of both sensors) for the initialization, but the data acquisition is not working correctly. Data from both sensors coming out the same value, although the sensors are not in the same "position".

### Your workbench
I am using ESP32 dev kit v4 (at 3V3) and two 1-axis sensors.
The connections are the 4 wires from the board to the sensors (VCC, GND, SDA, SCK) leaving unplugged the interrupt pins of the sensors, as I am using poll method)

### Steps to reproduce
My code is:

`#include <Wire.h>
#include "SparkFun_Displacement_Sensor_Arduino_Library.h"

ADS myFlexSensor1; //Create object of the ADS class
ADS myFlexSensor2; //Create object of the ADS class

void setup()
{
	Serial.begin(115200);
	Serial.println("SparkFun Displacement Sensor Example");

	Wire.begin(18, 19);

	//Setup first sensor - look for it at the default address of 0x13 = 19
	if (myFlexSensor1.begin(16) == false){
		Serial.println(F("First sensor not detected. Check wiring. Freezing..."));
		while (1);
	}

	//Setup second sensor - look for it at the I2C address of 45. You should have set this up in example 6
	if (myFlexSensor2.begin(21) == false){
		Serial.println(F("Second sensor not detected. Check wiring. Freezing..."));
		while (1);
	}

	delay(100);
}

void loop(){

	if (myFlexSensor1.available() == true && myFlexSensor2.available() == true){
		Serial.print("1:");
		Serial.print(myFlexSensor1.getX());
		Serial.print("\t2:");
		Serial.println(myFlexSensor2.getX());
	}
	else {
		Serial.println("...");
	}
	
	delay(100);
}`

### Expected behaviour
I believe I should see the two different values of the getX() functions

### Actual behaviour
The output on the serial terminal is:
`Port closed
Uploading to I/O board
Opening port
Port open
SparkFun Displacement Sensor Example
1:-28.15	2:-72.05
1:-88.05	2:-85.31
1:-81.16	2:-80.18
1:-80.18	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97
1:-80.97	2:-80.97`

Andlit, 2019-12-18:
I have exactly the same problems. I wanted to use multiple 1-axis sensors for a sensor glove. I had some discussions with Carlson from Bemdlabs. At the end he could not help me. Their library do not support multiple sensors, therefore I tried the Sparkfun libraries and examples. The libraries from Sparkfun are much advanced compared to the Bendlabs ones. But as described, more than one sensor do not work properly. They influence one another. I placed now this abstract in the Sparkfun Forum. This seems to be the only way to have support from Sparkfun engineers. Have you found other help?
Many regards Andreas

sideridispas, 2019-12-18:
@Andlit I managed to get them to work together but only by using the Wire library and parts of their original library (read register & filtering functions etc). Also, due to my project requirements, I wanted to read measurements via polling method and utilize only the 4 necessary pins, instead of all 6:
1. VCC
2. GND
3. SDA
4. SCL

The code that worked for me is the following:

```
#include <Wire.h>
#define ADS_TRANSFER_SIZE 3
#define SENSORS_NUMBER 5

typedef enum{
	ADS_RUN = 0,
	ADS_SPS,
	ADS_RESET,
	ADS_DFU,
	ADS_SET_ADDRESS,
	ADS_POLL,
	ADS_GET_FW_VER,
	ADS_CALIBRATE,
	ADS_AXES_ENABLED,
	ADS_SHUTDOWN,
	ADS_GET_DEV_ID
} ADS_COMMAND_T;

//Identifier for packet received from ADS
typedef enum{
	ADS_SAMPLE = 0,
	ADS_FW_VER,
	ADS_DEV_ID
} ADS_PACKET_T;

//I2C Pins
const int SCLpin = 16;
const int SDApin = 15;

//I2C Sensor Addresses
uint8_t address[] = {1,2,3,4,5};

uint8_t buffer[3];

volatile float currentSample[SENSORS_NUMBER]; //This is where the calculated angular values are stored pre and post filtering. 2 FINGERS NOW
static float prev_sample[SENSORS_NUMBER];
static float filter_samples[SENSORS_NUMBER][6];

void setup(){
	Serial.begin(115200);
	Serial.println("Bendlabs Custom I2C Lib");

	//I2C bus init
	Wire.begin(SCLpin, SDApin);
	Wire.setClock(100000);	

	//Sensors init check connection
	
	buffer[0] = ADS_POLL;//init polling mode config
	buffer[1] = true;

	for (int i = 0;i < SENSORS_NUMBER;i++) {
		Wire.beginTransmission(address[i]);
		if (Wire.endTransmission() != 0) {
			Serial.print("Problem with Sensor: ");Serial.println(i+1);
		}else {
			writeBuffer(buffer, ADS_TRANSFER_SIZE, address[i]);
			delay(10);
			Serial.print("OK Sensor: ");Serial.println(i + 1);
		}
	}
	Serial.println("Sensors Init complete!");
}

void loop(){
	for (int i = 0;i < SENSORS_NUMBER;i++) {
		if (readBuffer(buffer, ADS_TRANSFER_SIZE, address[i]) == true) {
			if (buffer[0] == ADS_SAMPLE) {
				currentSample[i] = (float)ads_int16_decode(&buffer[1]) / 64.0f;
				processNewData(i); //Run new samples through the deadband and signal filters
				Serial.print(i+1);Serial.print(": ");
				Serial.println(currentSample[i]);
			}
		}else {
			Serial.println("Data Problem");
		}
	}
	Serial.println();
	delay(100);
}

bool readBuffer(uint8_t *buffer, uint8_t len, uint8_t address){
	Wire.requestFrom(address, len);
	uint8_t i = 0;
	while (Wire.available()){
		if (i < len) {
			buffer[i++] = Wire.read();
		}
	}
	if (i == len) {
		return true;
	}
	return false; //Error
}

bool writeBuffer(uint8_t *buffer, uint8_t len, uint8_t address){
	Wire.beginTransmission(address);
	uint8_t bytesWritten = Wire.write(buffer, len);
	Wire.endTransmission();
	if (bytesWritten == len) {
		return true;
	}
	return false;
}

inline int16_t ads_int16_decode(const uint8_t *p_encoded_data){
	return ((((uint16_t)(p_encoded_data)[0])) |	(((int16_t)(p_encoded_data)[1]) << 8));
}

// Low pass IIR filter
void signalFilter(uint8_t finger){
	filter_samples[finger][5] = filter_samples[finger][4];
	filter_samples[finger][4] = filter_samples[finger][3];
	filter_samples[finger][3] = (float)currentSample[finger];
	filter_samples[finger][2] = filter_samples[finger][1];
	filter_samples[finger][1] = filter_samples[finger][0];

	// 20 Hz cutoff frequency @ 100 Hz Sample Rate
	filter_samples[finger][0] = filter_samples[finger][1] * (0.36952737735124147f) - 0.19581571265583314f * filter_samples[finger][2] +
		0.20657208382614792f * (filter_samples[finger][3] + 2 * filter_samples[finger][4] + filter_samples[finger][5]);

	currentSample[finger] = filter_samples[finger][0];
}

// Deadzone filter
void deadzoneFilter(uint8_t finger){
	float dead_zone = 0.5f;
	if (fabs(currentSample[finger] - prev_sample[finger]) > dead_zone) {
		prev_sample[finger] = currentSample[finger];
	}
	else {
		currentSample[finger] = prev_sample[finger];
	}
}

//Takes the data from the latest sample and loads it into the filters
void processNewData(uint8_t finger){
	// Low pass IIR filter
	signalFilter(finger);

	// Deadzone filter
	deadzoneFilter(finger);
}
```
As you can see, I have already configured the addresses to each sensors with the numbers 1, 2, 3, 4 & 5, before running this code. So you need to change the addresses too, before testing my code.

Hope that will help you :)

Andlit, 2019-12-18:
Hello Sideridispas,
many thanks for this program code. It is exactly the preferred method via polling to collect the data of the sensors. I will give you feedback soon (after christmas). 
Many regards
Andreas

nseidle, 2020-01-22:
Have you tried example #7 - dual sensor setup? It was designed to talk to two sensors at different addresses. The example was written for two dual axis sensors but it's easy enough to delete the getY() calls.

coltonottley, 2020-05-05:
Hi I was able to replicate the issue mentioned here. The solution is to not use static data type in the deadzone and signal filter functions. Nate I can email you a correct version if you'd like.

Andlit, 2020-05-05:
Hi Colton,
I'm glad to hear from you. "My" Softwareingeneer made a solution, but we are interessed to see your version.
Please send the code.
Thank you and many regards
Andreas

coltonottley, 2020-05-05:
Hi Andlit, I sent you my version to your email. I believe these changes will be getting pulled into Github shortly.

nseidle, 2020-05-13:
Should be fixed in latest release. Please reopen if problems continue.

closed 2020-05-13 by nseidle

Ashiq112, 2021-01-27:
I am trying to get angular readings from two one-axis flex sensors like "sideridispas". I am working with the example code " Dual Sensor Setup". I have changed the I2C address of my flex sensor, hooked them up to Adafruit bluefruit feather sense nrf52840,
and connected VCC, GND, SDA, SCK of the sensor and the nrf52840. 
The problem is the output shows the same readings on both sensor. Like:

1,40.31,	2,59.52,
1,61.01,	2,57.88,
1,56.35,	2,56.35,
1,56.35,	2,56.35,
1,56.35,	2,56.35,
1,56.35,	2,56.35,
1,56.35,	2,56.35,
1,56.35,	2,56.35,

But the senors are in different positions. I want to get independent data from individual sensors. I am using "SparkFun_Displacement_Sensor_Arduino_Library". 
 
The issue was closed here, but I am facing the same problem. Has the latest update been released?