sparkfun/SparkFun_Displacement_Sensor_Arduino_Library #7: Where can I get stretching data?

Korzhak, 2022-01-18:
### Subject of the issue
I worked with bendlabs' library for this sensors. This library gives me 2 kind of data from one-axis sensor: bending and stretching. But in your library I can't to give stretching data. Maybe I didn't find special function for that. Where can I find it?

### Your workbench
* Arduino platform
* ESP32 DEVKIT 1

### Expected behaviour
Can you tell me, how I can get this data or maybe you can give me some docs about your code. Thank you!

PS: my English is so bad, SORRY! I'm working on it!
Thank you very much!

nseidle, 2022-01-18:
Your english is very good! Thank you for reporting this issue. What library are you using from Bendlabs that supports the stretch measurement?

Our library is based on their work so I am not surprised we are trailing behind.

Korzhak, 2022-01-19:
Oh, sorry. I forgot to include link to library. I used [one_axis_ads](https://github.com/bendlabs/one_axis_ads) library. 
Examples [bend_stretch_polled_demo](https://github.com/bendlabs/one_axis_ads/tree/master/examples/bend_stretch_polled_demo) has such code in `loop`:
```C
static float sample[2];
int ret_val = ads_read_polled(sample, &data_type);

  // Check if read was successfull
  if(ret_val == ADS_OK)
  {
    if(data_type == ADS_SAMPLE)
    {
      // Low pass IIR filter
      signal_filter(sample);

      // Deadzone filter
      deadzone_filter(sample);

      Serial.print(sample[0]);    // Bend data
      Serial.print(",");
      Serial.println(sample[1]);  // Stretch data
    }
  }
```

This code give me `sample` array, and second value in this array is stretching data. 
I can try to change your library for getting stretching data if your library can't do this. But I'm not sure that I can do it right

nseidle, 2022-01-19:
Very interesting. I didn't know that had added that! Neat.

I very much welcome a pull request to add stretch support. Don't worry about doing it right :) I'll give you feedback on anything that does not look correct, but I'm sure you'll do fine.

Korzhak, 2022-01-21:
I found where is the problem. You wanted to do library for one and two axis sensors simultaneously. But original libraries for one- and two-axis have different approach to getting data from sensors.
One axis library has extended type `ADS_PACKET_T` (included `ADS_STRETCH_SAMPLE`. This parameter use for processing received stretching data) and `ADS_READ_STRETCH` instead `ADS_AXES_ENABLED` (this parameter exist in two_axis_library on the same place with 8 number). So for one-axis sensors you can use `ADS_AXES_ENABLED` for enabling stretching data and in method `available()` of ADS class you have to add one more `else if` for processing it.

I wrote methods in ADS class for enabling stretching. And I added some code for process this data. 
How I can push it to new branch?

Korzhak, 2022-01-21:
I fixed it and did pull request

closed 2022-01-28 by Korzhak