sparkfun/SparkFun_Displacement_Sensor_Arduino_Library #14: How to change I2C address

jendy12345, 2023-06-08:
Hello,I have some questions.May I ask?
I have purchased the product with part number 100080101-01-EVAL, which is listed as part number 2190-100080101-01-EVAL-ND. I encountered an issue where I need to change the sensor address. For example, I want to change the original sensor address from 18 to 30. Initially, I was able to successfully update the address, but after connecting via Bluetooth and powering off, the previously set sensor address reverted back to the default value and I am unable to change the address anymore. What could be the cause of this issue? How should I proceed if I still need to change the address?

I have encountered the same issue with both the Bluetooth module on the Bluno Beetle v1.1 and the HC-06 Bluetooth module connected to the Arduino UNO board. After modifying the address, it reverts back to the default address and cannot be changed again. Does this sensor support address modification? Currently, I have two devices with the address set to 18, but I need to modify three different addresses: 18, 30, and 40. How should I proceed to handle this?

I'm using this code to change the address. However, when I try to modify it, it doesn't change successfully in one attempt. I need to repeatedly refresh and modify it to make the change successful.I used the Example6_SetI2CAddress code file provided to make the modification.

[image]

Another issue I encountered is that I can successfully change the address when I set it to 18, but any other address setting fails.
It is possible that the sensor's address modification is locked or restricted？？

[image]

nanofortran, 2023-07-10:
I was able to successfully change addresses and then revert to the original address. One thing that tripped me up, may not be your issue, is that the program asks for input in decimal . . . the first few times I put 12 in thinking it was 0x12. Then I figured out what the program wanted. You can also look at the repository from BendLabs that the SparkFun library is based on here: https://github.com/bendlabs. I have been jumping between the two libraries: the SparkFun superior for ease of use, the BendLabs for understanding what is going on under the hood, so to speak. The i2c address change example works in the BendLabs repository as well.

Xiangyu-Fu, 2024-01-23:
I encountered the same problem with the two-axis sensor. Have you already solved it? No matter what I try, I can't change the address from 0x12 to any other address. From the code, I guess the address 0x12 may be the bootloader address of the two-axis sensor.